kevinakasam/KlackEnder-Probe #2: Marlin

probonopd, 2022-07-19:
Hello, this looks pretty genius. I especially like that you don't need to rewire the motherboard but can use the existing endstop cable (and possibly re-use the existing endstop switch?).

> A kind user (Farva) on my [Discord](https://discord.gg/xqpKrxt9FC) figured out how to configure Marlin for the Rev2. I had no time to test it yet. Feel free to join my server and test it out on your own. Once I tested it, I will upload detailed instruction here.

Is there any update on this? Could you post/link to that firmware configuration please?
Thanks, and keep up the great work.

kevinakasam, 2022-07-19:
Hey and thanks!
I just finished my exams last week and now I'm trying to work on all the request I got over the last months. This is also on my To-Do list. I still had no time to test the marlin config, but you can join my Discord and you will get some help with that :)

Farva42, 2022-07-21:
The Read me files for both Rev 1 & Rev 2 have been updated to include the recent changes to Marlin to better support this type of probe. I'll be working to include config files in the next little bit. If you have any issues I'm on discord regularly and can help.

closed 2022-07-21 by Farva42

probonopd, 2022-07-22:
Thank you very much. So nice to see this included in upstream Marlin.

What is the CODE needed to deploy the magnetic probe, do 9-point autoleveling, and put the magnetic probe back before starting the actual printing?

Farva42, 2022-07-22:
Start g-code would be 
G29 L0; load mesh saved in slot 0. 
G29 J3; probe 9 points and tilt loaded mesh. 

Before you print you will want to create your mesh. You can find the option in the menus under motion UBL tools. 
The probe will be automatically deployed any time there is a G29 command to probe as well as any time you ask it to home z if using rev 2.