kevinakasam/KlackEnder-Probe #12: Probe_calibrate not working so well

ishay320, 2023-02-14:
i used this macro:
```
[gcode_macro calibrate_nozzle]
gcode:
  G28
  PROBE_OUT
  G0 X117 Y117 # the center of print bed
  Probe_calibrate
  PROBE_IN
```

it does good on `G28` and on `probe_out`, when it get to probe calibrate it starts good but when it returns the probe it move to x0 y-8.
there is a reasen to move it to 0 0 on `probe_in`?

ishay320, 2023-02-14:
the fix that i have got is:
```
[gcode_macro calibrate_nozzle]
gcode:
  G28
  PROBE_OUT
  G0 X117 Y117 # the center of print bed
  Probe_calibrate
```
and removing the probe manually

kevinakasam, 2023-02-17:
> there is a reasen to move it to 0 0 on probe_in?

Yes because after the probe got stored the print may start and starting at the zero position might help with other macros you use for a purgeline or so

closed 2023-12-31 by oldcrazyeye