kevinakasam/KlackEnder-Probe #13: bed mesh macro

ishay320, 2023-02-15:
better macro for bed mesh:
```
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: _BED_MESH_CALIBRATE
description: Perform Mesh Bed Leveling with klicky automount
gcode:
	G90
    PROBE_OUT

    _BED_MESH_CALIBRATE {% for p in params
           %}{'%s=%s ' % (p, params[p])}{%
          endfor %}

    PROBE_IN
```

it works with the original bed mesh button

DasBurninator, 2023-02-16:
Its not just the bed mesh macro that has this issue that could be easily resolved. It is all of the macros. z_tilt has this issue as well.

I suggest using the klickyprobe macros instead. Much better written.

kevinakasam, 2023-02-17:
Thanks for the comment! Yes you're right renaming the existing macros makes the usage a little easier. I will give your macro a try when I'm back home and edit the Klackender.cfg.

oldcrazyeye, 2023-12-31:
No progress in this thread, closing

closed 2023-12-31 by oldcrazyeye