kevinakasam/KlackEnder-Probe #14: [Feature Request] Use klickyprobe macros

DasBurninator, 2023-02-16:
Instead of using the macros included that aren’t really written well, utilize the existing work of the Klicky Probe macros.

This would allow for a much better user experience.

The existing macros could be updated to be much better if the `rename_existing:` function was utilized.

kevinakasam, 2023-02-17:
Hey!
There is something in the making to get the Klicky macros working with the Klack. Harry from my Discord spent some time to get all the changes done. I will link this in the instructions soon so you can chose if you want to use my very basic macros or the more advanced Klicky-Klack macros:

https://github.com/Harrypulvirenti/Klack-Probe-Macros

DasBurninator, 2023-02-21:
I ran a diff on all of this... Its just renamed macros and the addition of one to handle manual screw adjust. Also removed z-tilt adjust which is more useful than manual screw adjust for some people. Why not just provide steps for setting up klickyprobe macros? This seems like overkill and just another set of macros to maintain vs just providing the config parts needed in the variables file.

kevinakasam, 2023-06-19:
Hey @Ordspilleren That would be great!
Feel free to add a Klicky Macro adaption block to the readme and then do a pull request :) Would be happy to merge it then