kevinakasam/KlackEnder-Probe #24: Please help Klackender Config issue

iiiiiiiiddddddkkkkkkk, 2023-08-14:
I just set up the klackender config and i got this error from klipper any suggestions?

Section 'menu __main' is not a valid config section

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

YungChin95, 2023-08-14:
Remove the bottom part of the config named "Klackender Menu"

iiiiiiiiddddddkkkkkkk, 2023-08-14:
ok i'll try that thanks for the rapid response

iiiiiiiiddddddkkkkkkk, 2023-08-14:
did that now i get this 

Option 'points' is not valid in section 'homing_override'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

iiiiiiiiddddddkkkkkkk, 2023-08-14:
> Remove the bottom part of the config named "Klackender Menu"

I did that now I get this error

Option 'points' is not valid in section 'homing_override'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

iiiiiiiiddddddkkkkkkk, 2023-08-14:
never mind i forgot to comment out the stuff in z tilt

closed 2023-12-31 by oldcrazyeye