kevinakasam/KlackEnder-Probe #27: relative_reference_index in the bed_mesh section is deprecated

eduhoribe, 2023-09-24:
Klipper deprecated the `relative_reference_index` option for the `bed_mesh` in the version `20230619`, details on how to update the configuration are specified [here](https://www.klipper3d.org/Bed_Mesh.html#the-deprecated-relative_reference_index).

After a quick read, my two cents is that we could replace it with the `zero_reference_position` pointing to one of the bed leveling screws.

miroan81, 2023-09-30:
I've just updated to the newest version and this thing popped out. After looking and reading I decided just to comment RRI and leave it like that, did bed mesh few times nothing weird happened, I guess zero_reference_position is not actually needed if everything works ok

oldcrazyeye, 2023-12-31:
Updated in the config now, thank you

closed 2023-12-31 by oldcrazyeye