kevinakasam/KlackEnder-Probe #38: HELP

yagoporto, 2024-06-22:
Good morning! I need your help, I did the entire installation of klackEnder, everything works normally, but when the switch is activated the nozzle continues to descend, it seems that it is not regular, I have already done the test with a multimeter and everything is working. 

I'm making an Ender 3 S1.

Follow my configuration file.

I didn't quite understand how to configure the "pin" in the probe part.

[probe]
pin: PA8 #Change to where you plug your probe in, endstop or probe pin pulled high (^)
z_offset: 0 #Measure per your specific setup
x_offset: 4 # negative = left of the nozzle
y_offset: 21 # negative = in front of of the nozzle
speed: 5.0
lift_speed: 15.0
sample_retract_dist: 1
samples: 2
samples_tolerance_retries: 6

The pin is correct?

Walkb4uRoll, 2024-07-03:
hey boss could you help compile for a ender 3 v2?

j0e57, 2024-08-15:
> I'm making an Ender 3 S1.
> 
> I didn't quite understand how to configure the "pin" in the probe part.
> 
> [probe] pin: PA8 #Change to where you plug your probe in, endstop or 
> The pin is correct?

I just try to help pin is where you attach your klackender cable.. its depend on your mainboard layout, if you use stock (maybe Creality 4.22 or 4.27 board) or other like SKR 1.3 1.4 SKR mini etc, its all use different pin for Z probe
According to https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h
Z probe pin is PA7 :) thats why nozzle continue to descend :)